cms-externals/geant4 #10: New Geant4 version 10.2p02

civanch, 2016-06-29:
A new Geant4 patch is available 10.2p02. It is needed to create a new branch in cms-externals/geant4.

The Geant4 distribution page: http://cern.ch/geant4/support/download.shtml

cmsbuild, 2016-06-29:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @smuzaffar, @degano, @davidlt, @Dr15Jones can you please review it and eventually sign/assign? Thanks.

cms-bot commands are list here cms-sw/cmssw#13029

civanch, 2016-06-29:
@smuzaffar , @degano , @iahmad-khan , I make an issue instead of PR because likely have no permission to create new branches.

civanch, 2016-07-03:
@smuzaffar , @degano , @iahmad-khan , @davidlange6, how we can progress with this request?

degano, 2016-07-03:
@iahmad-khan ping!

iahmad-khan, 2016-07-04:
@civanch the version updates  for data files, from the source page are as follows:

Versions updates for data files:
eant4-G4NDL 4.4                              ---> G4NDL4.5
geant4-G4EMLOW 6.35                    ---> G4EMLOW6.48
geant4-G4PhotonEvaporation 3.0     ---> G4PhotonEvaporation3.2
geant4-G4RadioactiveDecay 4.0      ---> G4RadioactiveDecay4.3.2
geant4-G4ENSDFSTATE 1.2.2         ---> G4ENSDFSTATE1.2.3

Do we need update these files as well for new patch?

civanch, 2016-07-04:
@iahmad-khan , you are right! This is the time when we should update data as well. May be we should be careful with G4NDL4.5 which is not used by CMS in production but may be larger than G4NDL4.4. Also may be these data already exit within list of externals but old one are linked to the DEVEL branch.

iahmad-khan, 2016-07-04:
@civanch in gcc530 branch it is the same. table updated for gcc530. Can you figure out , what out of these files ,  we need to update , so that i can do it accordingly. thanks.

civanch, 2016-07-04:
@iahmad-khan , for consistency we need update all files as you have identified.

closed 2016-07-18 by iahmad-khan