cms-externals/geant4 #16: Create a new branch cms/4.10.03.p01

civanch, 2017-04-04:
A new branch for Geant4 is needed.

cmsbuild, 2017-04-04:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed here cms-sw/cmssw#13029

davidlt, 2017-04-04:
Should be done now: https://github.com/cms-externals/geant4/tree/cms/4.10.03.p01
It now matches plain "4.10.03.p01" (imported from http://cern.ch/geant4/support/source/geant4.10.03.p01.tar.gz)

Thus it does not have changes from the previous GEANT4 branches.

civanch, 2017-04-04:
@davidlt , thanks for the code! How we should update data files? Type the list of modified files in this issue?

davidlt, 2017-04-04:
data files like G4NEUTRONXS?

https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_9_1_X/gcc700/geant4-G4NEUTRONXS.spec
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_9_1_X/gcc700/geant4-data-rpm.file

Those are fetched from the official source and we never had any custom modifications IIRC.

What would you like to change?

civanch, 2017-04-04:
Yes, for this release we need to copy to CMS data area several new datasets:
   G4EMLOW6.50
   G4PhotonEvaporation4.3.2
   G4RadioactiveDecay5.1.1
   G4ENDSDFSTATE2.1
Please, note, that the plan is to update to this new Geant4 in the DEVEL branch (not in 9_1_X), we will need change spec configurations for the DEVEL branch when these new datasets will be integrated into CMSSW.

smuzaffar, 2017-04-04:
@civanch , we can update geant4 data fils along wth geant4 in devel IBs. I would suggest to make a cmsdist PR along with geant4 and G4data changes.

civanch, 2017-04-04:
Yes, I will do a PR when #18183 will be accepted (this is a resolution of a small inconsistency between current CMSSW and newer G4). Without this it will not compile.

Concerning data, I suspect that when making cmsdist PR data the files should be already copied to the list of externals. Or it can be done simultaneously?

smuzaffar, 2017-04-04:
the data can be updated simultaneously. All you need is to update the geant4-*.spec files (e.g geant4-G4EMLOW.spec) in the same PR.

closed 2017-05-30 by smuzaffar