cms-externals/geant4 #19: Make a new branch for Geant4 10.4

civanch, 2017-07-07:
It was agreed to start examine Geant4 10.4 for possible use in 2018. For that a new branch is needed. It is planned update it after each Geant4 reference tag. The initial version of Geant4 is available as 10.4beta release: http://geant4.web.cern.ch/geant4/support/download_beta.shtml

cmsbuild, 2017-07-07:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2017-07-07:
@iahmad-khan , can you please create cms/4.10.04.b01 branch based on geant4 tar file mentioned in above link?

iahmad-khan, 2017-07-07:
@civanch , the new branch is here , https://github.com/cms-externals/geant4/tree/cms/4.10.04.b01 , do you need to put something on top of it before we update it in cmsdist ?

civanch, 2017-07-07:
@iahmad-khan , thanks , I just will try compile against it to see if there are issues in CMSSW sub-libraries or not (I expect nothing or minor problems), after that a change for cmsdist for the ROOT branch may be done. Will notify when ready.

This version of Geant4 strongly requires change of data files G4EMLOW6.54 and G4PhotonEvaporation5.0, which should be added to externals as well.

civanch, 2017-08-29:
Is done

closed 2017-08-29 by civanch