cms-externals/geant4 #21: Update Geant4 version for development branches

civanch, 2017-08-29:
We need update ROOT6 development branch with the new reference tag of Geant4 - new PR #19 is created. No change in data files.

For the DEVEL branch needs to be updated for a new patch release 10.3p02. It is available in GitHub: https://github.com/Geant4/geant4/releases  
No change in data files for this patch compared with 10.3p01

cmsbuild, 2017-08-29:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2017-09-14:
@smuzaffar,  to complete synchronising Geant4 ref-08 also extra data set G4EMLOW7.1 should be integrated to CMS data files. This modification will not affect standard CMS simulation but may affect it next next reference tags. The tar is available at afs: ~vnivanch/public/G4EMLOW7.1.tar.gz

smuzaffar, 2017-09-19:
@mrodozov , can you please take care of it.

smuzaffar, 2017-09-19:
@civanch , G4EMLOW is updated  https://github.com/cms-sw/cmsdist/pull/3426
By the way, we took the G4EMLOW.7.1.tar.gz dircetly from http://geant4.cern.ch/support/source/G4EMLOW.7.1.tar.gz

closed 2017-09-19 by smuzaffar

civanch, 2017-09-19:
@smuzaffar , @mrodozov , Thanks!