cms-externals/geant4 #23: Switch DEVEL branch to Geant4 10.3p02

civanch, 2017-09-19:
Geant4 10.3p02 includes several technical fixes important for MT run with many threads. This patch is available at https://github.com/Geant4/geant4/releases 

No change of Geant4 data files is needed. This Geant4 version should be added only to the DEVEL branch.

cmsbuild, 2017-09-19:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2017-09-21:
assign externals

cmsbuild, 2017-09-21:
New categories assigned: externals

@smuzaffar,@mrodozov you have been requested to review this Pull request/Issue and eventually sign? Thanks

smuzaffar, 2017-09-21:
thanks @mrodozov

closed 2017-09-21 by smuzaffar