cms-externals/geant4 #24: Switch ROOT6 branch to the new reference version of Geant4 10.3ref09

civanch, 2017-10-02:
A new reference version of Geant4 is available 10.3ref09: ~vnivanch/public/geant4-10-03-ref-09.tar.gz

This version may be directly installed into 10.4beta branch and is needed for ROOT6 IB only. No new data files are provided, only source code.

cmsbuild, 2017-10-02:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

mrodozov, 2017-10-02:
@civanch , the new version is available in the https://github.com/cms-externals/geant4/commits/cms/4.10.04.b01 branch and the test could be followed here https://github.com/cms-sw/cmsdist/pull/3471

civanch, 2017-10-02:
@mrodozov , thanks!

closed 2017-10-10 by civanch