cms-externals/geant4 #27: Create a branch for Geant4 10.4

civanch, 2017-12-13:
A new version of Geant4 is available. Because it is a new reference version which may be used for 2018 a new branch for it would be needed. It will be first used in 10_0_ROOT6 IB for testing.

The code is available: ~vnivanch/public/geant4-10-04/geant4.10.4.tar.gz
or
~vnivanch/public/geant4-10-04/geant4

It is needed to take the source from this place and not from Geant4 official git repository, because 3 bug fixes are added on top.

cmsbuild, 2017-12-13:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2017-12-14:
@mrodozov , can you please take care of it

mrodozov, 2017-12-15:
The new branch is here https://github.com/cms-externals/geant4/tree/cms/4.10.04
I'm testing it locally and then it will go on cmsdist as PR.
M.

mrodozov, 2018-04-17:
I'm closing this

closed 2018-04-17 by mrodozov