cms-externals/geant4 #35: New branch for validation of Geant4 10.5

civanch, 2018-09-19:
The tar file with the development version of Geant4 is available at url: 
https://cernbox.cern.ch/index.php/s/phtIQEYQc6HbOGn

Extra data files are required from /cvmfs/geant4.cern.ch/share/data :
1) G4NEUTRONXS2.0
2) G4EMLOW7.7
 
The new version of Geant4 will be used for validation for long time, so we will need and separate CMSSW development branch.

cmsbuild, 2018-09-19:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2018-09-24:
@civanch , the link is not working. Can you please put the geant4 tar file in some public some (afs or on any cmsdev machine)?

mrodozov, 2018-09-26:
We tried building it using the latest vecgeom commit available: 
https://gitlab.cern.ch:8443/VecGeom/VecGeom/commit/6a52158ad0e46407c8cb710e0159f449069506b3
and it failed with:
```
In file included from /build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/solids/specific/include/G4UPolycone.hh:46:0,
                 from /build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/solids/specific/include/G4Polycone.hh:70,
                 from /build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/divisions/include/G4ParameterisationPolycone.hh:47,
                 from /build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/divisions/src/G4PVDivision.cc:43:
/build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/management/include/G4UAdapter.hh: In instantiation of 'class G4UAdapter<vecgeom::cxx::UnplacedPolycone>':
/build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/solids/specific/include/G4UPolycone.hh:57:28:   required from here
/build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/management/include/G4UAdapter.hh:74:38: error: no members matching 'vecgeom::cxx::UnplacedPolycone::operator delete' in 'cl\
ass vecgeom::cxx::UnplacedPolycone'
     using UnplacedVolume_t::operator delete;
                                      ^~~~~~
/build/mrodozov/builds/geant/build_geant/BUILD/slc6_amd64_gcc700/external/geant4/10.04.ref07a/geant4.10.04.ref07a/source/geometry/management/include/G4UAdapter.hh:75:38: error: no members matching 'vecgeom::cxx::UnplacedPolycone::operator new' in 'class\
 vecgeom::cxx::UnplacedPolycone'
     using UnplacedVolume_t::operator new;
```

smuzaffar, 2018-09-26:
@mrodozov thanks.
@civanch , do we need to use some special version of VecGeom to build new geant4? Currently we have vecgeom v00.05.00 in cms externals. Mircho tried updating it to tip of master branch but geant4 still fails with the error mentioned above.

civanch, 2018-09-26:
@mrodozov , @smuzaffar , would it be possible to disable VecGeom in this branch? It is known, that an updated version of VecGeom is needed for this version of Geant4 but in reality there is no good version of VecGeom for this particular version of Geant4. It will be available later in the next iteration.

smuzaffar, 2018-09-27:
@civanch , we can drop vecgeom for geant4 but it is directly used in cmssw too ( https://github.com/cms-sw/cmssw/search?q=vecgeom&unscoped_q=vecgeom ) , so no idea what will happen is vecgeom and geant4 both are loaded in one process?

By the way, while building this version of geant4 I see that it mentions the follow data packages too which are not part of our distribution, do we need to bundle these too?
```
  G4PII (1.3)
  RealSurface (2.1.1)
  G4INCL (1.0)
```

civanch, 2018-09-27:
@smuzaffar , Experts promise to provide VecGeom ~ in 2 weeks time just after migration of Geant4 development from svn to git. This is nearly achieved but exact date is not known. So, if it is possible to build a branch without VecGeom, so Geant4 will be built without VecGeom, VecGeom will not compile.
If it is too complicate we are waiting for few weeks.

Data files are a separate thing, which may be done now. I would propose to integrate RealSurface and G4INCL, which potentially may be used (not used now). G4PII will not be needed - 100% sure.

smuzaffar, 2018-09-27:
thanks, cmsdist PR https://github.com/cms-sw/cmsdist/pull/4380 includes the geant4 update (with vecgeom dependency removed). I will test it against 10.3.GEANT4.X IB (once available).