cms-externals/geant4 #36: Switch GEANT4 branch to Geant4 10.5

civanch, 2018-12-12:
A new Geant4 10.5 is available ; https://github.com/Geant4/geant4/releases 

This new version requires new data files:
G4NDL4.5          - old dataset
G4EMLOW7.7
G4PhotonEvaporation5.3,
G4RadioactiveDecay5.3
G4SAIDDATA2.0
G4PARTICLEXS1.1  - new external
G4ABLA3.1,    - old dataset
G4INCL1.0
G4ENSDFSTATE2.2 - old dataset

cmsbuild, 2018-12-12:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2019-02-12:
done long ago

closed 2019-02-12 by civanch