cms-externals/geant4 #37: New Geant4 10.4patch03 version

civanch, 2019-02-12:
New Geant4 10.4p03 is available

in github: https://github.com/Geant4/geant4/releases/tag/v10.4.3
in gitlab: https://gitlab.cern.ch/geant4/geant4-dev/tree/geant4-10-04-patch-03 
at download page: http://geant4.web.cern.ch/support/download_archive 

No change in data files.

It is expected to be used for Run-2 legacy simulation.

cmsbuild, 2019-02-12:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2019-02-16:
@smuzaffar , new branch name likely should be cms/10.04.p03

fabiocos, 2019-02-16:
We have agreed with PPD/PdmV to have a parallel build CMSSW_10_5_0_pre2_g4104p3 to test this new version vs the present setup

smuzaffar, 2019-02-18:
@civanch , we now have https://github.com/cms-externals/geant4/commits/cms/v10.4.3 based on https://github.com/Geant4/geant4/releases/tag/v10.4.3 . I cherry-pick only two of our changes (needed for  from https://github.com/cms-externals/geant4/commits/cms/4.10.04 
- https://github.com/cms-externals/geant4/commit/f5a27d9b0a47f2a40fc27db06a3851685005bf72
- https://github.com/cms-externals/geant4/commit/7788cfbecfee824344bd3761cb32efe2bdc53046
I think every thing else has been integrated in up stream.
Please let me know if this looks good so that I can go ahead with cmsdist update

civanch, 2019-02-18:
@smuzaffar , thanks! these "atomic" modifications are unavoidable. I think we can go forward.

civanch, 2019-02-19:
done

closed 2019-02-19 by civanch

smuzaffar, 2019-02-19:
@civanch , by the way, last night we had built an IB ( CMSSW_10_5_GEANT104_X_2019-02-18-2300 ) with this version of geant4. Every thing looks good https://cmssdt.cern.ch/SDT/html/cmssdt-ib/#/ib/CMSSW_10_5_X for this IB. It is available for slc7_amd64_gcc700 (in case you want to run any additional tests).

civanch, 2019-02-19:
@smuzaffar , thanks, a simple test is running. We with Sunanda will run also physics tests, this will need more time.