cms-externals/geant4 #38: Switch GEANT4 branch from 10.5 to recent reference tag 10.5ref03

civanch, 2019-04-06:
@smuzaffar , a new reference tag of Geant4 10.5ref03 is available: https://gitlab.cern.ch/geant4/geant4-dev/tree/geant4-10-05-ref-03 

If this link does not work I will make a tar file. 

It is expected, that performance with updated Geant4 will be a bit changed. We need to make a new branch 10.5ref03 in cms-external/geant4 and after switch GEANT4 IB to this version. No change in data files are needed.

cmsbuild, 2019-04-06:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

fabiocos, 2019-04-07:
@civanch this is of course meant for 11_0_X, not for UL reprocessing, right?

smuzaffar, 2019-04-08:
@civanch , I can not access  https://gitlab.cern.ch/geant4/geant4-dev repository. Can you please make a tar file.
@fabiocos , for now we are only going to add this in special GEANT4 IBs.

civanch, 2019-04-08:
@fabiocos , this is even may be after 11_0 - it is a rolling validation of Geant4 developments.

civanch, 2019-04-08:
@smuzaffar , please, try out : https://cernbox.cern.ch/index.php/s/d86g9lrqpbR1kOD