cms-externals/geant4 #39: Switch Geant4 branch to Geant4 10.6beta

civanch, 2019-06-30:
A new Geant4 public version is available : https://github.com/Geant4/geant4/releases 
We need to use this new version in GEANT4 special IB for early testing.

This beta-release require modifications in other external packages:
New dataset G4PARTICLEXS2.0: http://geant4.web.cern.ch/support/download_beta 
New CLHEP version 2.4.1.2: http://proj-clhep.web.cern.ch/proj-clhep/clhep23.html
New VecGeom: https://gitlab.cern.ch/VecGeom/VecGeom/tree/v1.1.2

cmsbuild, 2019-06-30:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

mrodozov, 2019-07-10:
@civanch the IB is building and will be available later today, I'm closing this as resolved

closed 2019-07-10 by mrodozov