cms-externals/geant4 #40: Switch GEANT4 branch to the new reference version Geant4 10.5ref08

civanch, 2019-09-02:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-05-ref-08
This Geant4 version includes main development for 2019.
We need to use this new version in GEANT4 special IB for early testing.

This beta-release require modifications in other external packages:
New dataset G4PARTICLEXS2.1: ~vnivanch/public/G4PARTICLEXS2.1
New VecGeom: https://gitlab.cern.ch/VecGeom/VecGeom/tree/v01.01.04

cmsbuild, 2019-09-02:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2019-09-18:
Done a week ago and is working fine.

closed 2019-09-18 by civanch