cms-externals/geant4 #41: Switch GEANT4 branch to the new reference version Geant4 10.6cand00

civanch, 2019-11-12:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-06-cand-00/geant4 or
https://cernbox.cern.ch/index.php/s/Tn5QBVWxnY3hwEc 

This Geant4 version is nearly 10.6 release, which is scheduled for December,6.
We need to use this new version in GEANT4 special IB for early testing in order to feedback.

This release require modifications in other external datasets:
G4EMLOW7.9 : https://cernbox.cern.ch/index.php/s/sRJD69a7wNwR51Y 
RadioactiveDecay5.4 : https://cernbox.cern.ch/index.php/s/p6O7bR0yYbNtIzv
PhotonEvaporation5.4 : https://cernbox.cern.ch/index.php/s/9XIxLDyJre3EyiJ
G4NDL4.6 : https://cernbox.cern.ch/index.php/s/5E8480vJsfvnQyg

civanch, 2019-11-12:
@smuzaffar , @bsunanda, please have a look.

cmsbuild, 2019-11-12:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2019-11-12:
assign externals

cmsbuild, 2019-11-12:
New categories assigned: externals

@smuzaffar,@mrodozov you have been requested to review this Pull request/Issue and eventually sign? Thanks

smuzaffar, 2019-11-12:
+externals
A new branch cms/4.10.06cand00 has been created using ~vnivanch/public/geant4-10-06-cand-00/geant4 . PR https://github.com/cms-sw/cmsdist/pull/5344 includes these changes.

closed 2019-11-12 by smuzaffar

cmsbuild, 2019-11-12:
This issue is fully signed and ready to be closed.

bsunanda, 2019-11-12:
I believe one need to use VecGeom version 1.1.4 rather than 0.5.0 which we have been using in the standard builds.

smuzaffar, 2019-11-12:
@bsunanda , GEANT4 IBs are already using VecGeom v1.1.4

bsunanda, 2019-11-12:
That is great. Can you let me know the name of the build?