cms-externals/geant4 #42: Switch GEANT4 branch to Geant4 10.6

civanch, 2019-12-09:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-06/geant4 

We need to use this new version in GEANT4 special IB.

This release require modifications in other external datasets:
PhotonEvaporation5.5 : ~vnivanch/public/PhotonEvaporation5.5
CLHEP version 2.4.1.3
VecGeom v1.1.5

cmsbuild, 2019-12-09:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@davidlange6, @Dr15Jones, @smuzaffar, @fabiocos, @kpedro88 can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2019-12-11:
@civanch , I can not access your ~vnivanch/public/geant4-10-06/geant4 . Can we just get it from https://github.com/Geant4/geant4/releases or do you have any changes on top of it in your area?

civanch, 2019-12-11:
@smuzaffar , unfortunately, there are to fixes (pure numerical for rare division by zero and precision lost) on top of the release. For some reason all cmake and gmake related files have strong permissions in this directory. 
I will redo preparation  from scratch and let you know when ready.

smuzaffar, 2019-12-11:
I think it is better to create a new branch using official geant4 sources and then we can apply the patches via PR

smuzaffar, 2019-12-11:
@civanch , https://github.com/cms-externals/geant4/tree/cms/v10.6.0 is based on official geant4 tag https://github.com/Geant4/geant4/tree/v10.6.0 . Can you make a PR with additional changes? Or if yor provide me a tar.gz then I can do a diff and apply changes myself

fabiocos, 2019-12-11:
I agree it is better to have a clean start, and track whatever addition on top of that in a transparent way

civanch, 2019-12-11:
@smuzaffar , yes, I agree with this, indeed this would be more transparent.

smuzaffar, 2019-12-12:
@civanch , do you have your updates? As a wrote, just provide me a tar with and I can make a PR

civanch, 2019-12-12:
@smuzaffar , thanks, I will try making PR, simply missed that the branch is already there.

smuzaffar, 2019-12-18:
this has been integrated in geant4 external repo now. once cms-sw/cmssw#28637 is merged then we can move forward with geant4 10.6 integration in CMSSW GEANT4 IBs.

closed 2019-12-18 by smuzaffar