cms-externals/geant4 #44: Update GEANT4 IB with Geant4 10.6 patch01

civanch, 2020-02-14:
The first patch to Geant4 10.6 is available: https://github.com/Geant4/geant4/releases 

It require  update of data files G4EMDATA, which can be taken from

  ~vnivanch/public/G4EMLOW7.9.1

This is Geant4 version we expect in future to merge to the master. For the time being it needs to be added to GEANT4 IB only.

cmsbuild, 2020-02-14:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @smuzaffar, @silviodonato, @makortel, @davidlange6, @fabiocos can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2020-02-18:
geant4 10.6.1 is now here https://github.com/cms-externals/geant4/tree/cms/v10.6.1 . CMs fix https://github.com/cms-externals/geant4/pull/43 ( https://github.com/cms-externals/geant4/pull/43 ) has already been merged upstream. So we do not have to patch it any more.

We will update cmsdist to integrate it in GEANT4 IBs.

smuzaffar, 2020-02-18:
cmsdist PR to integrate 10.6.1 is https://github.com/cms-sw/cmsdist/pull/5562

closed 2020-02-18 by smuzaffar