cms-externals/geant4 #47: New Geant4 release 10.7beta for the GEANT branch

civanch, 2020-06-29:
New Geant4 release 10.7beta is available. It includes ongoing Geant4 developments toward 10.7 version to be released in December.  It may  be uploaded from url:

https://github.com/Geant4/geant4/releases

It is expected to use this only in GEANT4 branch. It is also needed to update VecGeom package with vecgeom 1.1.7

cmsbuild, 2020-06-29:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @silviodonato, @dpiparo, @smuzaffar, @makortel can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2020-06-30:
@mrodozov , can you please take care of it?

smuzaffar, 2020-07-01:
this has been integrated for GEANT4 Ibs

closed 2020-07-01 by smuzaffar