cms-externals/geant4 #48: Switch GEANT4 branch to Geant4 10.6ref09

civanch, 2020-10-12:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-06-ref-09
This Geant4 version includes main development for 2020.
We need to use this new version in GEANT4 special IB for early testing.

cmsbuild, 2020-10-12:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

mrodozov, 2020-10-21:
is this change to be put on the 10_6 branch:
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/g4_106/geant4.spec
or on the branch that is currently using geant4 10.7 and vecgeom 1.1.7 ? 
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/geant4/geant4.spec

civanch, 2020-10-21:
@mrodozov , only on the branch with geant4 10.7 and vecgeom 1.1.7.
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/geant4/geant4.spec

mrodozov, 2020-10-21:
So this is 10.7 release then, and I should better put it's name 10.7.something , or I keep the branch name 10.6.9 ?

civanch, 2020-10-21:
10.6.9 is more precise - this is used by the G4 team.

mrodozov, 2020-10-21:
Understood. It's clear now. Thank you

closed 2020-10-22 by smuzaffar