cms-externals/geant4 #49: Switch GEANT4 IB to Geant4 10.6ref10

civanch, 2020-10-29:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-06-ref-10
This Geant4 version includes main development for 2020.
We need to use this new version in GEANT4 special IB for early testing.

This time we would ask to build GEANT4 IB without usage of VecGeom.

cmsbuild, 2020-10-29:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2020-11-02:
@mrodozov , @smuzaffar , we need to test new Geant4 version in order to be able promptly feedback Geant4. In parallel, this version if built without VecGeom will help testing DD4Hep - it is even more urgent.

mrodozov, 2020-11-02:
is this change to be put on
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/g4_106/geant4.spec
or on the branch that is currently using geant4 10.6.9 ?
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/geant4/geant4.spec

civanch, 2020-11-02:
@mrodozov , on the last one substituting 10.6.9. What is essential for DD4Hep validation is to disable VecGeom.

smuzaffar, 2020-11-02:
@mrodozov , this shoudl go in to https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_2_X/geant4/geant4.spec branch (where we currently have geant4 10.6.9)

mrodozov, 2020-11-02:
Yes, this is clear now. 
@civanch , since when the cmake flag 
-DGEANT4_USE_USOLIDS="all" 
was expecting VecGeom should be available, I'd assume it shouldn't be used.
When I used it it searched for vecgeom and the config fails.
Instead I used 
-DGEANT4_USE_USOLIDS-ADVANCED:INTERNAL=1
and it builds , but I can't find example cmake configs on github or gitlab .
Do you know more on that, is this a right way to be build ? 
I remember something for Usolids being a substitute for VecGeom, so one or the other should always be used, 
but some of the Usolids options are expecting VecGeom. There should be a way to use Usolids alone, correct ?

mrodozov, 2020-11-04:
@civanch we have now this changes integrated in the last IB, 
https://cmssdt.cern.ch/SDT/html/cmssdt-ib/#/relVal/CMSSW_11_2/2020-11-03-2300?selectedArchs=slc7_amd64_gcc820&selectedFlavors=GEANT4_X&selectedStatus=failed
14 relvals failing (looks like due to the changes)

civanch, 2020-11-04:
@mrodozov , I am in contact with geometry experts of Geant4. This problem is known in Run-3 geometry, which we already fixed but not yet propagated to GT. But this should not appears in Run1 and Run2 WFs.

civanch, 2020-11-04:
@mrodozov , I am running this new IB and it is working fine to me no problems, no crash. I do not yet understand if VecGeom is used or not used. It is a contradiction to what happens in standard IB WFs.

mrodozov, 2020-11-04:
@civanch the geant4 is built without vecgeom but I see I had to remove the vecgeom references from the toolfiles for scram

mrodozov, 2020-11-08:
@civanch after https://github.com/cms-sw/cmsdist/pull/6370 the last IB doesn't include vecgeom references anyhow.

civanch, 2020-11-08:
@mrodozov , thanks!

closed 2020-11-08 by civanch