cms-externals/geant4 #50: Switch GEANT4 branch to the new version 10.7cand00

civanch, 2020-11-09:
A new Geant4 reference version is available : ~vnivanch/public/geant4-10-07-cand-00
This Geant4 version is very close to the final 10.7, which is our target for the Run-3.
We need to use this new version in GEANT4 special IB for early testing. we have ~2 weeks when we still have a chance to feedback Geant4 if any problem will be discovered.

Additionally to the code new data files were updated:
~vnivanch/public/G4PARTICLEXS3.1
~vnivanch/public/G4EMLOW7.13
~vnivanch/public/ENSDFSTATE2.3
~vnivanch/public/PhotonEvaporation5.7
~vnivanch/public/RadioactiveDecay5.6
~vnivanch/public/RealSurface2.2

Would it be possible to provide two IBs: one with VecGeom, another - without VecGeom?

cmsbuild, 2020-11-09:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2020-11-11:
@mrodozov , @smuzaffar , this version of Geant4 is very similar to the final 10.7, data files will not be changed anymore, so such migration will cover most part of the migration to our target version 10.7. It would be beneficial if we will have such IBs .

mrodozov, 2020-11-11:
we can use the current config for the IB without and then switch to the old one for an IB with VecGeom.
In short yes I'll just open two prs to be merged

smuzaffar, 2020-11-11:
@civanch , is there any geant4 in their official github repo for this? Also looks like data files are not yet available on geant4 download site. How can we make sure that the data files we get from your home area will be the one they will upload to their site?

civanch, 2020-11-11:
@smuzaffar , datafiles will be public when 10.7 will be released. I do not expect they will be different. The only question which is not clear to me: I may provide link to cvmfs of geant4 and not copy data structure to my afs. What is better?

Concerning code itself - there is a branch in git. In past I tried to provide a link to a branch but you have no permission to take it.

When 10.7 will be release - 4 weeks from now all links will be available.

smuzaffar, 2020-11-16:
@civanch , we now have two IBs for this
- G4VECGEOM_X : with VecGeom
- GEANT4: wiout vecgeom
11h00 Ibs for both of these flavors building and will soon be available for use.

closed 2020-11-16 by smuzaffar