cms-externals/geant4 #51: Switch to Geant4 10.7

civanch, 2020-12-04:
New version of Geant4 is available: https://github.com/Geant4/geant4/releases/tag/v10.7.0 

Migration to this version will be done in two steps: 1st include into CMSSW_11_3_GEANT4_X and into CMSSW_11_3_G4VECGEOM_X, after validation it will be proposed for CMSSW_11_3_X.

New version of external tools are needed:
1) CLHEP 2.4.4.0: https://proj-clhep.web.cern.ch/DISTRIBUTION/
2) VecGeom v01.01.08: https://gitlab.cern.ch/VecGeom/VecGeom/-/tree/v01.01.08

cmsbuild, 2020-12-04:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2020-12-05:
@mrodozov , @smuzaffar , this Geant4 version 10.7 is practically the same as one we already integrated 10.7cand00. The main difference - I give links to official git repository instead of making private copy. New CLHEP and VecGeom are minor bug fix releases.

civanch, 2020-12-07:
@mrodozov , @smuzaffar , Just now a problem was identified: in build specification an extra directory for includes should be added:  $GEANT4CORE_BASE/include/ (until now all includes were in $GEANT4CORE_BASE/include/Geant4). This is a bug, which will be fixed in the first patch.

mrodozov, 2020-12-07:
I see something breaking in that compilation job not sure what yet, although I built the external first to be sure it builds. 
So from what I understand this means this line
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_11_3_X/geant4_vecgeom/geant4-toolfile.spec#L43
should not be $GEANT4CORE_BASE/include/Geant4 but we have to temporarily change it to $GEANT4CORE_BASE/include
only. Correct ?

civanch, 2020-12-07:
@mrodozov , we need both places for includes.

mrodozov, 2020-12-17:
The two IBs are built, closing

closed 2020-12-17 by mrodozov