cms-externals/geant4 #52: Switch to Geant4 10.7p01

civanch, 2021-02-05:
The first patch to Geant4 10.7 is available: https://github.com/Geant4/geant4/releases/tag/v10.7.1 

This patch includes many mainly technical fixes, which should not change physics. One fix the build of PLT sub-library build, now there is no need anymore to add to path a special place for includes.

Migration to this version should starts from including into  CMSSW_11_3_GEANT4_X with native geometry and also into CMSSW_11_3_G4VECGEOM_X with VecGeom.

Additionally to the source code, a new version of data set G4PARTICLEXS3.1.1  is needed, which can be taken from url https://geant4.web.cern.ch/support/download

cmsbuild, 2021-02-05:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

mrodozov, 2021-02-09:
should go to geant4 with vecgeom (note for myself)

civanch, 2021-02-09:
CMSSW_11_3_GEANT4_X branch - native geometry
CMSSW_11_3_G4VECGEOM_X - with VecGeom

For CMSSW_11_3_0_pre3_G410.7.1 IB the version with VecGeom

civanch, 2021-02-14:
It is done.

closed 2021-02-14 by civanch