cms-externals/geant4 #53: Switch to Geant4 10.7p02

civanch, 2021-06-14:
The Geant4 team provide a new patch 10.7p02:  https://geant4.web.cern.ch/support/download 

This patch includes mainly technical fixes, so it is possible first check on regression the master. If the regression will be broken, then this version should go some testing to GEANT4 and G4VECGEOM branches first.

cmsbuild, 2021-06-14:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @dpiparo, @silviodonato, @smuzaffar, @makortel, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

mrodozov, 2021-06-17:
@civanch you want me to try it against cmsdist/master, if it works we get it as it is, if it doesn't we first put it in 

https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_12_0_X/geant4
https://github.com/cms-sw/cmsdist/blob/IB/CMSSW_12_0_X/geant4_vecgeom

first to be fixed and then to cmsdist/master when it's fixed. Did I got it correct ?

civanch, 2021-06-17:
@mrodozov , if we see no regression when compare with the master, this will not means that this patch is wrong or buggy, it simply will require more complicate statistics test. We should start from geant4 and geant4_vecgeom IBs and after will ask a validation campaign.

civanch, 2021-06-17:
@mrodozov , yes, the master and geant4_vecgeom are identical for some time since we updated master to the previous Geant4 tag. I agree, that 10.7p02 should be built in this branch. If last time this was built long ago we may not expect regression but we should not worry about too much.

civanch, 2021-06-21:
#7041 and #7042 are merged

closed 2021-06-21 by civanch