cms-externals/geant4 #55: Switch G4VECGEOM to Geant4 10.7ref09

civanch, 2021-11-08:
Now Geant4 team is working on the new release 11.0. In order to be better prepared would it be possible to try to include Geant4 reference version and make a new G4VECGEOM IB ?

Geant4 10.7ref09 can be taken from: /afs/cern.ch/user/v/vnivanch/public/geant4.11.0/Geant4.10.7.9.tar.gz

There is a difficulty connected with DD4Hep, which also depend on Geant4 and it is not known if the current DD4Hep will compile on top of this version of Geant4 or not. CMSSW is already modified to be compatible with old and new Geant while DD4Hep not yet.

cmsbuild, 2021-11-08:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @makortel, @smuzaffar, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2021-11-08:
@ddaina , can you please take care of this?

ddaina, 2021-11-09:
FYI: https://github.com/cms-sw/cmsdist/pull/7436

closed 2022-02-04 by civanch