cms-externals/geant4 #56: Switch G4VECGEOM and GEANT4 branches to Geant4 11.0

civanch, 2022-02-04:
We need to migrate Geant4 branches to the new Geant4 version 11.0.0 :
https://github.com/Geant4/geant4/releases 

This version required update of the data files: https://geant4.web.cern.ch/support/download 
G4EMLOW8.0 and G4PARTICLEXS4.0

cmsbuild, 2022-02-04:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @makortel, @smuzaffar, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2022-02-07:
@iarspider , can you please take care of it?

smuzaffar, 2022-02-07:
cmsdist branches where this should be updated are
- IB/CMSSW_12_3_X/geant4
- IB/CMSSW_12_3_X/geant4_vecgeom

smuzaffar, 2022-02-08:
@iarspider FYI

iarspider, 2022-02-09:
@smuzaffar on it.

smuzaffar, 2022-02-10:
thanks @iarspider 
@civanch  @bsunanda , today we have first 12.3.X (GEANT4 and G4VECGEOM) IBs with geant4 11.0.0 available

closed 2022-02-10 by smuzaffar