cms-externals/geant4 #57: Migration of Geant4 branches to Geant4 11.0.1

civanch, 2022-03-08:
There is the first patch to geant4 release 11.0: https://github.com/Geant4/geant4/releases 
Please, provide migration of GEANT4 and G4VECGEOM branches to this version of Geant4.

cmsbuild, 2022-03-08:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @makortel, @smuzaffar, @qliphy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

iarspider, 2022-03-09:
@smuzaffar on it.

smuzaffar, 2022-03-09:
@civanch  , following cmsdist PRs updates geant4 11.0.1 for special G4 IBs
- g4 with vecgeom: https://github.com/cms-sw/cmsdist/pull/7678
- g4: https://github.com/cms-sw/cmsdist/pull/7679

closed 2022-03-09 by smuzaffar