cms-externals/geant4 #61: Migration Geant4 branches to Geant4 11.0.7

civanch, 2022-08-30:
Geant4 11.0.7 reference version is available. Please update GEANT4 and G4VECGEOM branches. Code is available via: 
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.0.7.tag.gz

After that it would be interesting to update to the new DD4hep version v01-22 available on official url: https://github.com/AIDASoft/DD4hep

cmsbuild, 2022-08-30:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2022-08-31:
@iarspider @aandvalenzuela  can you please take care of it?

civanch, 2022-09-09:
Is done in https://github.com/cms-sw/cmsdist/pull/8060 and https://github.com/cms-sw/cmsdist/pull/8061

closed 2022-09-09 by civanch