cms-externals/geant4 #63: Migration Geant4 branches to Geant4 11.0.8

civanch, 2022-10-06:
Geant4 11.0.7 reference version is available. Please update GEANT4 and G4VECGEOM branches. Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.0.8.tag.gz 

The difference with the previous reference version geant4.11.0.7 is mainly technical, physics results should be unchanged. We requesting this update in order to see if any issue appears for CMSSW and to be able continue validation of coming public Geant4 release. No change in other externals or in data files is needed.

civanch, 2022-10-06:
@iarspider , this is a relatively simple update. I have prepared tar file privately, because it is an internal geant4 reference version.

iarspider, 2022-10-07:
@civanch the URL gives me 404 error, could you please check?

iarspider, 2022-10-07:
nvm, figured the right url.

civanch, 2022-10-07:
@iarspider , please, try again - I have changed permissions with chmod.

closed 2022-10-31 by civanch