cms-externals/geant4 #66: Migration Geant4 branches to Geant4 11.1

civanch, 2022-12-10:
New Geant4 version 11.1 is available: https://github.com/Geant4/geant4/releases 

We need to migrate GEANT4 and G4VECGEOM branches to this version. The result of CPU tests show degradation of CPU performance if Geant4 is compiled with -O2. It is proposed to compile Geant4 with -O3 but left VecGeom compiled with -O2. There should be no problem for dd4hep - it is used only at initialisation of SIM, so its CPU efficiency does not matter - it may be -O3 or -O2.

cmsbuild, 2022-12-10:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2022-12-11:
@civanch , https://github.com/cms-sw/cmsdist/pull/8226 (for G4 Vecgeom) and https://github.com/cms-sw/cmsdist/pull/8227 ( for Geant4) are open now.

smuzaffar, 2023-01-11:
@civanch , I am trying to setup a G4 LTO based IB so that we can build 13.0.0.pre3.GEANT4LTO pre-release. Currently we have two Geant4 based IBs

- Geant4 without Vecgeom
- Geant4 with Vecgeom

As master IBs/releases use Geant4 with Vecgeom so is it OK if we only setup Geant4 with Vecgeom+LTO IBs?

civanch, 2023-01-11:
@smuzaffar , yes, we should use G4VECGEOM as a baseline. Geant4 without VecGeom IB is used only for cross comparisons and not needed for the new validation campaign. What we need in 13.0.0.pre3.GEANT4LTO:
CLHEP, DD4HEP, VECGEOM, and Geant4 from the current G4VECGEOM IB +LTO.

closed 2023-01-18 by smuzaffar

smuzaffar, 2023-01-19:
@civanch , CMSSW_13_0_0_pre3_G4VECGEOM and CMSSW_13_0_0_pre3_LTOG4VECGEOM releases are available now. I think you need to open a validation campaign for these.