cms-externals/geant4 #69: Migration Geant4 branches to Geant4 11.1.1

civanch, 2023-02-15:
The 1st patch release for Geant4 11.1 is available: https://github.com/Geant4/geant4/releases 

Please, create a new branch cms/4.11.1  and use it for the new GEANT4 and G4VECGEOM IBs. The patch is mainly technical, we expect little change in results. No modification is needed in other related packages CLHEP, VecGeom, DD4hep.

We also would like to make a special IB CMSSW_12_6_0_G4VECGEOM for testing of simulation versus 2022 data.

This patch partially was included already inside cms/4.11.0 as CMS private patches, which were tested. Currently there is ongoing PdmV validation campaign 13_0_0_pre3_LTOG4VECGEOM.

civanch, 2023-02-15:
@bsunanda, FYI

cmsbuild, 2023-02-15:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2023-02-16:
@iarspider , it may be straightforward to make a new branch. Creation of a new IBs is the second step.

smuzaffar, 2023-02-17:
@civanch , do we need the changes you added on top of `11.1.0` ( https://github.com/cms-externals/geant4/compare/9f34590941fb8a3f7ad139731089ec3794947545...cms/v11.1.0 )  for new patch releases  `11.1.1` ?

civanch, 2023-02-17:
@smuzaffar , no, all my private commits to 11.1.0 are inside 11.1.1.

civanch, 2023-02-20:
@smuzaffar , thank you for new IBs, first checks are showing that it is OK. WOuld it be possible to make a special IB CMSSW_12_6_0_G4VECGEOM on top of 12_6 with this G4 11.1.1? It is needed for Sunanda tests of MC versus data.

smuzaffar, 2023-02-20:
@iarspider , can you please build this special build? We do not need any official build for this, just go to one of cmsdev nodes and build CMSSW_12_6_0_G4VECGEOM  using cmsdist of CMSSW_12_6_0 + geant4/vecgeom releases changes for G4 11.1.1.

bsunanda, 2023-02-21:
Could you keep me informed if that version (CMSSW_12_6_0_G4VECGEOM using cmsdist of CMSSW_12_6_0 + geant4/vecgeom releases changes for G4 11.1.1) is built. Thanks and regards

iarspider, 2023-02-21:
@civanch @bsunanda the special build of CMSSW_12_6_0_G4VECGEOM is available on `cmsdev24` under `/build/razumov/g4/BUILD`.

bsunanda, 2023-02-21:
Please let me know how I can access this. Can I do cmsrel command on cmsdev24for CMSSW_12_6_0_G4VECGEOM?

smuzaffar, 2023-02-21:
@bsunanda , just do the same as you did for the last build of G4 i.e.
```
# login to cmsdev24
# start container
cmssw-el8
# setup env for new area
source /build/razumov/g4/BUILD/cmsset_default.sh
# create dev area
scram p CMSSW_12_6_0_G4
# and then setup cmsenv and checkout/build/test your code
```

closed 2023-02-24 by civanch