cms-externals/geant4 #71: Migration Geant4 branch G4VECGEOM to Geant4 11.1.ref04

civanch, 2023-04-30:
Geant4 11.1.ref04 reference version is available. Please update G4VECGEOM branch. We need this to control Geant4 developments and to validate modifications for the next public patch to Geant4.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/g4.11.1.ref04.tag.gz

cmsbuild, 2023-04-30:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-05-02:
@civanch , I opened https://github.com/cms-sw/cmsdist/pull/8486 . Note that this is only for G4Vecgeom IBs. Do we need this to go in Geant4 IBs too?

smuzaffar, 2023-05-03:
done, next G4VEcgeom IBs will be based on Geant4 11.1.ref04

closed 2023-05-03 by smuzaffar