cms-externals/geant4 #72: Migration Geant4 branches to Geant4 11.1.2

civanch, 2023-06-19:
The 2nd patch release for Geant4 11.1 is available: https://github.com/Geant4/geant4/releases 

We would propose to integrate this patch with a goal to be used in production in 2024. The proposed fixes are mainly technical, physics should not be affected in a visible way, but simulation histories likely will be different.

Please, integrate this patch and use it in the G4VECGEOM and GEANT4 IBs, after testing of these IBs we will propose validation campaign in order to integrate this patch to CMSSW master.

cmsbuild, 2023-06-19:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-06-20:
@iarspider , can you please take care of it?

smuzaffar, 2023-06-20:
as we do not have any CMS specific changes for Geant4 11.1.1, so I would suggest to directly take the 11.1.2 from Geant4 github repo.

iarspider, 2023-06-21:
Completed via https://github.com/cms-sw/cmsdist/pull/8556 and https://github.com/cms-sw/cmsdist/pull/8557

closed 2023-06-21 by iarspider

smuzaffar, 2023-09-12:
@civanch , you want `11.1.2` G4 in the 13.3.X master IBs ... right? This is the version we have used to build `CMSSW_13_2_0_pre3_G4VECGEOM` release for validation