cms-externals/geant4 #73: Migration Geant4 branches to Geant4 11.1.ref07

civanch, 2023-07-30:
Geant4 11.1.ref07 reference version is available. Please update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.1.ref07.tag.gz

Also special build for work with low-pileup data of Run2 is needed. The standard receipt for that I cannot find out.

cmsbuild, 2023-07-30:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @perrotta, @dpiparo, @rappoccio, @makortel, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-08-01:
@iarspider , thanks for taking care of this. As requested during core sw meeting, once integrated in G4/VecG4 IBs , we also need to build 12.6 release

iarspider, 2023-08-04:
@civanch We have noticed that `G4persistency` library is not being built  - do we need to enable it explicitly?

smuzaffar, 2023-08-04:
@iarspider , previously geant4 was building [G4persistency](https://github.com/Geant4/geant4/blob/v11.1.2/source/persistency/CMakeLists.txt#L5) and now it has been renamed to [G4geomtext](https://github.com/cms-externals/geant4/blob/cms/v11.1.ref07/source/persistency/CMakeLists.txt#L3) and [G4mctruth](https://github.com/cms-externals/geant4/blob/cms/v11.1.ref07/source/persistency/CMakeLists.txt#L6) . I would suggest to replace `G4persistency` with `G4geomtext` and `G4mctruth` in cmsdist/scram tool files

smuzaffar, 2023-08-04:
and as `-DGEANT4_USE_GDML=ON` is set so also add [G4gdml](https://github.com/cms-externals/geant4/blob/cms/v11.1.ref07/source/persistency/CMakeLists.txt#L10) in toolfile

civanch, 2023-08-05:
is done

closed 2023-08-05 by civanch