cms-externals/geant4 #74: Migration Geant4 branches to Geant4 11.1.ref08

civanch, 2023-09-08:
Geant4 11.1.ref08 reference version is available. Please update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.1.ref08.tag.gz

Also special build for work with low-pileup data of Run2 is needed for Sunanda tests.

cmsbuild, 2023-09-08:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@Dr15Jones, @rappoccio, @smuzaffar, @makortel, @sextonkennedy, @antoniovilela can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-09-08:
@iarspider , can you please take care of it?

iarspider, 2023-09-11:
https://github.com/cms-sw/cmsdist/pull/8694, https://github.com/cms-sw/cmsdist/pull/8693

civanch, 2023-09-11:
@iarspider , @smuzaffar , thanks!

closed 2023-09-11 by civanch