cms-externals/geant4 #75: Migration Geant4 branches to Geant4 11.1.ref09

civanch, 2023-10-08:
Geant4 11.1.ref09 reference version is available. Please update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branch 11.1.ref08 is not needed anymore.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.1.ref09.tag.gz

We know for sure that between ref08 and ref09 nothing was changed in physics, so we will need to perform technical tests only. A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests but not for this case.

cmsbuild, 2023-10-08:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@sextonkennedy, @antoniovilela, @smuzaffar, @Dr15Jones, @rappoccio, @makortel can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-10-09:
@iarspider , can you please take care of it?

iarspider, 2023-10-09:
@civanch compared to ref08, the `G4VEmModel::SetLPMFlag()` method that is used by `g4hepem` is gone. Is it intentional?

civanch, 2023-10-10:
@iarspider , thanks for info! Please, reload tar from the same place - it is patched now.

iarspider, 2023-10-11:
@civanch thanks, restarted the tests.

iarspider, 2023-10-23:
Closed via #8741

closed 2023-10-23 by iarspider