cms-externals/geant4 #77: Migration Geant4 branches to Geant4 11.1.ref10

civanch, 2023-11-05:
Geant4 11.1.ref10 reference version is available. Please update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branch 11.1.ref09 is not needed anymore.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.1.ref10.tag.gz

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests but not for this case.

cmsbuild, 2023-11-05:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@antoniovilela, @smuzaffar, @Dr15Jones, @rappoccio, @makortel, @sextonkennedy can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2023-11-07:
@smuzaffar , is it possible to check this intermediate development version of G4?

smuzaffar, 2023-11-08:
@iarspider , can you please take care of it?

smuzaffar, 2023-11-13:
@civanch , this is now available in 14.0.X GEANT4 IBs

closed 2023-11-13 by smuzaffar

civanch, 2023-11-13:
@smuzaffar  , @iarspider , Thanks!