cms-externals/geant4 #78: Migration Geant4 branches to Geant4 11.2.cand02

civanch, 2023-11-20:
Geant4 11.2.cand02 reference version is available. Please, update G4VECGEOM and GEANT4 branches. This is the last check before the new Geant4 11.2 (planed for December,8). We need this test branches in order to have a last minute feedback to Geant4. The previous branch 11.1.ref10 is not needed anymore.

Code is available via:
http://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.2.cand02.tag.gz

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests but not for this case. We need only technical test of the candidate version.

cmsbuild, 2023-11-20:
A new Issue was created by  @civanch Vladimir Ivantchenko.

@antoniovilela, @makortel, @sextonkennedy, @Dr15Jones, @rappoccio, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2023-11-21:
@iarspider

smuzaffar, 2023-11-22:
done

closed 2023-11-22 by smuzaffar