cms-externals/geant4 #79: Migration Geant4 branches to Geant4 11.2.1

civanch, 2024-02-17:
The 1st patch to Geant4 11.2 is available: https://geant4.web.cern.ch/download/11.2.1.html 
We plane to validate this version in GEANT4 and G4VECGEOM branches of CMSSW and consider it for the CMS simulation production in 2025.

The regression 11.2.1/11.2.0 is not expected - simulation histories will be different. The physics itself is not changed and statistical agreement is assumed. Data files are also unchanged.

civanch, 2024-02-17:
assigned @iarspider

cmsbuild, 2024-02-17:
cms-bot internal usage

cmsbuild, 2024-02-17:
A new Issue was created by  @civanch.

@rappoccio, @sextonkennedy, @antoniovilela, @makortel, @Dr15Jones, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2024-02-19:
Thanks! Would it be possible to make also special build for @bsunanda tests as recently was done for 11.2.0?

iarspider, 2024-02-19:
Sure. Do you still need the 11.2.0 build?

civanch, 2024-02-19:
No, all results are already obtained.

closed 2024-02-20 by civanch

iarspider, 2024-02-21:
@civanch @bsunanda special version of CMSSW_12_6_0 - `CMSSW_12_6_0_G41121` - is available. To use it, login to cmsdev32 and execute the following commands:
```
$ cmssw-cc8
$ source /build/razumov/geant4_11.2.1/BUILD/cmsset_default.sh
$ scram list -c CMSSW # make sure only the newly-built CMSSW_12_6_0_G41121 is listed
$ scram p CMSSW_12_6_0_G41121
```

bsunanda, 2024-02-21:
Thanks

But I do not find any CMSSW project when I do the scram list command. I am getting
SCRAM warning: >>>> No SCRAM project CMSSW available for architecture el8_amd64_gcc10. <<<<
You can run "scram list" to see the available projects and their versions.

Please advise
Sunanda

iarspider, 2024-02-21:
@bsunanda Sorry for the typo - use `cmssw-cc7` to start the container, not `cmssw-cc8`.

bsunanda, 2024-02-21:
Thanks that works

Sunanda