cms-externals/geant4 #80: Migration Geant4 branches to Geant4 11.2.ref03

civanch, 2024-04-16:
Geant4 11.2.ref03 reference version is available. Please, update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branches with intermediate development versions 11.1.ref0X are not needed anymore. This one will be needed for 1-2 months.

Code is available via: geant4.11.2ref03.tar.gz](https://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.2ref03.tar.gz)

New dataset: [G4PARTICLEXS4.1.tar.gz](https://vnivanch.web.cern.ch/vnivanch/hadronic/G4PARTICLEXS4.1.tar.gz)

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests but not for this case.

cmsbuild, 2024-04-16:
cms-bot internal usage

civanch, 2024-04-23:
@smuzaffar , @iarspider , can you, please, take care?

closed 2024-05-31 by civanch