cms-externals/geant4 #81: Migration Geant4 branches to Geant4 11.2.ref05

civanch, 2024-05-31:
Geant4 11.2.ref05 reference version is available. Please, update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branches with intermediate development versions 11.2.ref0X are not needed anymore. This one will be needed for checks before end of June public beta-release. We do not plan migration to it inside mainstream CMSSW only for feedback.

Code is available at https://vnivanch.web.cern.ch/vnivanch/hadronic/geant4.11.2.ref05.tar.gz 

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests is also needed.

cmsbuild, 2024-05-31:
cms-bot internal usage

cmsbuild, 2024-05-31:
A new Issue was created by  @civanch.

@sextonkennedy, @smuzaffar, @rappoccio, @antoniovilela, @makortel, @Dr15Jones can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

closed 2024-06-06 by civanch