cms-externals/geant4 #82: Migration Geant4 branches to Geant4 11.2.2

civanch, 2024-06-27:
A new patch of the next the first Geant4 11.2.2 is available: 
https://geant4-dev.web.cern.ch/download/11.2.2.html

This incluses technical  fixes, which does no change of physics. It expected to migrate from version 11.2.1  to 11.2.2 in the master after validations within G4VECGEOM and GEANT4 branches. Both 11.2.1 and 11.2.2 will be used and both will be kept for long.

A special build for work with low-pileup data of Run2 is needed for Sunanda tests.

The branch 11.2ref03 and 11.2ref05 likely will not be used future.

cmsbuild, 2024-06-27:
cms-bot internal usage

cmsbuild, 2024-06-27:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

closed 2024-07-02 by civanch