cms-externals/geant4 #83: Migration Geant4 branches to Geant4 11.2.ref06

civanch, 2024-07-24:
Geant4 11.2.ref06 reference version is available (it is equivalent to Geant4 11.3beta). Please, update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branches with intermediate development versions 11.2.ref0X are not needed anymore. We do not plan migration to ref06 mainstream CMSSW only provide feedback.

Code is available at https://ivanchenko.web.cern.ch/vnivanch/hadronic/geant4.11.2.ref06.tar.gz

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests is also needed.

cmsbuild, 2024-07-24:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

iarspider, 2024-07-24:
@civanch the link you provided doesn't work.

civanch, 2024-08-22:
It turned out, that the new feature introduced in Geant411.3beta contradict to RunManager of CMSSW. So, we close this request and will open new when a new Geant4 reference version will be available.

closed 2024-08-22 by civanch