cms-externals/geant4 #85: Migration Geant4 branches to Geant4 11.2.ref08

civanch, 2024-10-03:
Geant4 11.2.ref08 reference version is available together with new data files. Please, update G4VECGEOM and GEANT4 branches. We need this to control Geant4 developments. The previous branches with intermediate development versions 11.2.ref07 is not needed anymore. We do not plan migration to ref08 mainstream CMSSW only provide feedback.

Code is available at https://ivanchenko.web.cern.ch/hadronic/geant4.11.2.ref08.tar.gz

New datasets are available at the same location:
https://ivanchenko.web.cern.ch/hadronic/G4EMLOW8.6.1.tar.gz
https://ivanchenko.web.cern.ch/hadronic/G4ENSDFSTATE3.0.tar.gz
https://ivanchenko.web.cern.ch/hadronic/PhotonEvaporation6.0.tar.gz
https://ivanchenko.web.cern.ch/hadronic/RadioactiveDecay6.0.1.tar.gz

A special build for work with low-pileup data of Run2 for Sunanda tests is also needed.

cmsbuild, 2024-10-03:
cms-bot internal usage

cmsbuild, 2024-10-03:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2024-10-04:
@iarspider , can you please take care of it

closed 2024-10-08 by civanch

smuzaffar, 2024-10-17:
This is now available in 14.2.X GEANT4/G4VECGEOM IBs.
@iarspider , feel free to go ahead with building special 12.6 build