cms-externals/geant4 #86: Migration Geant4 branches to Geant4 11.3.cand00

civanch, 2024-11-09:
Geant4 11.3.cand00 reference version is available together with new data files. Please, update G4VECGEOM and GEANT4 branches. We need to check this version to control candidate variant of Geant4 11.3 planned for December, 2024. The previous branches with intermediate development versions 11.2.ref08 is not needed anymore.

The code is available at url: https://ivanchenko.web.cern.ch/geant4.11.3.cand00.tar.gz

New data sets:
https://ivanchenko.web.cern.ch/PhotonEvaporation6.1.zip 
https://ivanchenko.web.cern.ch/RadioactiveDecay6.1.zip

cmsbuild, 2024-11-09:
cms-bot internal usage

cmsbuild, 2024-11-09:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

civanch, 2024-11-11:
@smuzaffar , @iarspider , we basically have 1 week to make feedback to this Geant4 release. If something will go wrong after we would be able to propose fixes only for patches to 11.3.

smuzaffar, 2024-11-11:
@civanch , I have opened https://github.com/cms-sw/cmsdist/pull/9513

smuzaffar, 2024-11-14:
@civanch , this has been integrated in to our G4 IBs.

closed 2024-11-14 by smuzaffar