cms-externals/geant4 #87: Migration Geant4 branches to Geant4 11.3

civanch, 2024-12-06:
The new Geant4 11.3 version is available together with new data files. Please, update G4VECGEOM and GEANT4 branches. We may potentially consider this version for 2025. The previous branches with intermediate development versions 11.3.cand00 are not needed anymore.

The code is available at url: https://geant4.web.cern.ch/download/11.3.0.html 

An extra data file is needed:  https://cern.ch/geant4-data/datasets/G4RadioactiveDecay.6.1.2.tar.gz with the most recent data for the radioactive decay.

A special build for work with low-pileup data of Run2 for Sunanda tests is also needed - we will do full set of validations.

cmsbuild, 2024-12-06:
cms-bot internal usage

cmsbuild, 2024-12-06:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2024-12-06:
@iarspider , can you please take care of it

civanch, 2025-02-18:
+1

Thanks!