cms-externals/geant4 #88: Migration Geant4 branches to Geant4 11.3ref01

civanch, 2025-02-18:
The new Geant4 11.3ref01 version is available. Please, update G4VECGEOM and GEANT4 branches. We may potentially consider this version for 2025. 

The code is available at url: https://cernbox.cern.ch/s/xjhtKCkOtMyNhUw 

We will use this temporary version for tests only.

We need speacial IB for 2016 low-pileup data analysis.

cmsbuild, 2025-02-18:
cms-bot internal usage

cmsbuild, 2025-02-18:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2025-02-21:
@civanch , this has been integrated in to Geant4 IBs.

> We need speacial IB for 2016 low-pileup data analysis.

@iarspider , I think SIM also need special 12_6 build using this geant4 version

iarspider, 2025-02-21:
> [@civanch](https://github.com/civanch) , this has been integrated in to Geant4 IBs.
> 
> > We need speacial IB for 2016 low-pileup data analysis.
> 
> [@iarspider](https://github.com/iarspider) , I think SIM also need special 12_6 build using this geant4 version

Sure, I will prepare it.

civanch, 2025-02-25:
@iarspider , thanks!

closed 2025-02-25 by civanch

smuzaffar, 2025-02-26:
@civanch , with GCC 14 we see many warnings like [a] coming from geant4. It is coming from https://github.com/cms-externals/geant4/blob/cms/v11.3.ref01/source/geometry/management/src/G4VSolid.cc#L218-L234 when For Union Solid both `touchesA and touchesB` are not `true` https://github.com/cms-externals/geant4/blob/cms/v11.3.ref01/source/geometry/solids/Boolean/src/G4UnionSolid.cc#L207-L212 . Can you please check if this is already fixed in geant4 upstream? If not then can some default values be set for min/max X/Y/Z  [here](https://github.com/cms-externals/geant4/blob/cms/v11.3.ref01/source/geometry/management/src/G4VSolid.cc#L221) ? 
 

[a] https://cmssdt.cern.ch/SDT/cgi-bin/buildlogs/el8_amd64_gcc14/CMSSW_15_1_X_2025-02-25-2300/BigProducts/Simulation
```
geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc: In member function 'GetCubicVolume':
 geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:206:21: warning: 'minX' may be used uninitialized [-Wmaybe-uninitialized]
 In member function 'EstimateCubicVolume',
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4BooleanSolid.cc:427:39,
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4UnionSolid.cc:578:47:
  geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:229:32: warning: 'maxX' may be used uninitialized [-Wmaybe-uninitialized]
 geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc: In member function 'GetCubicVolume':
geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:206:26: note: 'maxX' was declared here
  geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:206:31: warning: 'minY' may be used uninitialized [-Wmaybe-uninitialized]
 In member function 'EstimateCubicVolume',
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4BooleanSolid.cc:427:39,
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4UnionSolid.cc:578:47:
  geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:230:32: warning: 'maxY' may be used uninitialized [-Wmaybe-uninitialized]
 geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc: In member function 'GetCubicVolume':
geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:206:36: note: 'maxY' was declared here
In member function 'EstimateCubicVolume',
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4BooleanSolid.cc:427:39,
    inlined from 'GetCubicVolume' at geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/solids/Boolean/src/G4UnionSolid.cc:578:47:
  geant4/11.2.2-fb614397ac47af8bc49d5b94628e5f70/geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:231:32: warning: 'minZ' may be used uninitialized [-Wmaybe-uninitialized]
 geant4.11.2.2/source/geometry/management/src/G4VSolid.cc: In member function 'GetCubicVolume':
geant4.11.2.2/source/geometry/management/src/G4VSolid.cc:206:41: note: 'minZ' was declared here
```