cms-externals/geant4 #89: Migration Geant4 branches to Geant4 11.3ref02

civanch, 2025-03-10:
The new Geant4 11.3ref02 version is available. Please, update G4VECGEOM and GEANT4 branches. We do not consider this version for the master.

The code is available at url: https://cernbox.cern.ch/s/vxSDZZgruf5sQnO

We will use this temporary version for tests only. Previous version 11.3.ref01 is already tested and we will not work with it anymore.
We need speacial IB for 2016 low-pileup data analysis.

We expect that the problem reported in https://github.com/cms-externals/geant4/issues/88 is fixed.

cmsbuild, 2025-03-10:
cms-bot internal usage

cmsbuild, 2025-03-10:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

iarspider, 2025-03-11:
@civanch one of the files in that archive - `examples/extended/electromagnetic/TestEm1/r.out` - exceeds GitHub's limit of 100 MB and can't be stored in git repository (and therefore I can't start the tests for this new version). I guess this file is not needed and can be removed?

civanch, 2025-03-11:
@iarspider , sorry about that. This file not needed as well as full directory examples.

iarspider, 2025-03-11:
https://github.com/cms-sw/cmsdist/pull/9742