cms-externals/geant4 #92: Migration of Geant4 branches to Geant4 11.3ref04

civanch, 2025-05-02:
The new Geant4 11.3ref04 version is available. Please, update G4VECGEOM and GEANT4 branches. We do not consider this version for the master.

The code is available at url:  

We will use this temporary version for tests only. Previous version 11.3.ref03 is already tested and we will not work with it anymore.
We need special IB for 2016 low-pileup data analysis.

cmsbuild, 2025-05-02:
cms-bot internal usage

cmsbuild, 2025-05-02:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

iarspider, 2025-05-05:
I have started building " special IB for 2016 low-pileup data analysis", should be available soon.

iarspider, 2025-05-07:
`CMSSW_12_6_0_G4113ref04` is now available.