cms-externals/geant4 #95: Migration of Geant4 branches to Geant4 11.3ref07

civanch, 2025-08-02:
The new Geant4 11.3ref07 version is available. Please, update G4VECGEOM and GEANT4 branches. We do not consider this version for the master. We do not need updating G4ADEPT branch with this Geant4.

The code is available at url: https://cernbox.cern.ch/s/PJFZwDcQdN4GlHx

We will use this temporary version for tests only. Previous version 11.3.ref05 is already tested and we will not work with it anymore. 11.3.ref06 is used in the branch G4ADEPT, it should be used longer for this branch.

We need special IB for 2016 low-pileup data analysis.

cmsbuild, 2025-08-02:
cms-bot internal usage

cmsbuild, 2025-08-02:
A new Issue was created by  @civanch.

@Dr15Jones, @antoniovilela, @makortel, @mandrenguyen, @rappoccio, @sextonkennedy, @smuzaffar can you please review it and eventually sign/assign? Thanks.

cms-bot commands are listed <a href="http://cms-sw.github.io/cms-bot-cmssw-issues.html">here</a>

smuzaffar, 2025-08-04:
@iarspider , can you please open cmsdist PR to update geant4?

civanch, 2025-08-04:
@smuzaffar , I have no strong preference, 11.3ref06 and 11.3ref07 are similar. There may be only one reason to use some stable Geant4 version - reproducibility of problems in AdePT. I am not sure if we will need this approach for long.

civanch, 2025-08-04:
You are right! I will try to make PR to the master, if it will work.

smuzaffar, 2025-08-04:
@civanch , while checking difference between ref5, ref6 and ref7 I noticed thatfor ref6 we have the following change [a] i.e. update G4EMLOW to 8.7. We did not update this g4 data for ref6. Should we update it for current G4 IBs and Adept IBs ( which also have ref6) ?

[a]
```
--- g4-ref5/cmake/Modules/G4DatasetDefinitions.cmake    2025-08-04 13:01:42.252856557 +0200
+++ g4-ref6/cmake/Modules/G4DatasetDefinitions.cmake    2025-08-04 13:00:19.402032892 +0200
@@ -18,11 +18,11 @@
 # - Low energy electromagnetics
 geant4_add_dataset(
   NAME      G4EMLOW
-  VERSION   8.6.1
+  VERSION   8.7
   FILENAME  G4EMLOW
   EXTENSION tar.gz
   ENVVAR    G4LEDATA
-  MD5SUM    9db67a37acc3eae9b0ffdace41a23b74
+  MD5SUM    949c9422ae668208562be1b991750df1
   )

```

civanch, 2025-08-04:
@smuzaffar , you are right, we may update this dataset. I did not forced this update, because 8.6.1 -> 8.7 is an addition of the data, which CMS will never use. But it is useful; we will not need update later.

smuzaffar, 2025-08-05:
@iarspider , can you please update G4EMLOW geant4 data for g4 and gh4adept branch too?  It should be a separate PRs with just the g4 data update

civanch, 2025-10-02:
Thanks, it is done and tested.

closed 2025-10-02 by civanch